sqfmi/watchy-hardware #2: STEP models won't open in Solidworks 2020

Snowflake6, 2021-01-13:
[image]

Same result with Watchy.step and Watchy_battery.step

sqfmi, 2021-01-13:
We tried opening the files just now and it appears ok:

[image]

We Googled the error you saw and found the following solutions:

https://www.cati.com/blog/2019/03/bypassing-warning-encountered-invalid-record-index-during-step-import/
https://www.javelin-tech.com/blog/2018/11/encountered-invalid-record-index-solidworks/

Let us know if these solve the problem for you, we are not familiar with Solidworks so we were unable to reproduce the error

Snowflake6, 2021-01-13:
Well, that was truly strange.  I loaded it on my home computer, same Solidworks license, and it opened up.  I'll try again when I get to the office tomorrow but it appears to work here as well.  Thanks!  I think this can be closed for now.
[image]

closed 2021-01-13 by Snowflake6