sqfmi/watchy-hardware #3: Missing 32KHz oscillator for Low-Power BLE connectivity

LeonMatthes, 2021-02-02:
As mentioned in [this issue](https://github.com/espressif/esp-idf/issues/947 ), BLE with light sleep on the ESP32 is currently only supported with the use of an external 32KHz oscillator.

Unfortunately, Watchy currently does not have this, which is going to make it very difficult to maintain a true low power Bluetooth connectivity to a smartphone. For me this is a major use case for a smart watch...

Even more unfortunate is that the RTC can provide the exact waveform needed at its 32KHz output pin, however, it's not connected to the right pin of the ESP (GPIO 32_XP). And the pin is already used for BTN_3. (At least I think, I'm not a Hardware expert).

Maybe this could be fixed in a new revision of the Watchy hardware (perhaps even in the crowd supply campaign :pray: ).

m986883511, 2021-02-08:
I also have this problem，我用的的是现成的ESP32-WRROOM-32模块，没法加32k的晶振，这个是个不错的想法，希望得到乐鑫的解答。

LeonMatthes, 2021-02-10:
After looking at the ESP32 Hardware reference manual, I just wanted to add that the input pin for the 32KHz waveform might have to be the 32_XN (IO33) pin, and 32_XP might need to be connected to ground through a capacitor, but I'm not certain on this.
It would just be amazing to see the 2mA power consumption with BLE active that can be seen in the prior thread!

Jan-NiklasB, 2021-03-25:
If the devs don't see any problems with the changes, this should fix it: #5

Jan-NiklasB, 2021-03-25:
NVM, have to do it over again, LeonMatthes is right with the assumption that the source has to be connected to 32K_X**N** and a capacitor has to be placed between GND and 32K_XP...

sqfmi, 2021-03-25:
@Jan-NiklasB also it is recommended the signal amplitude be 0.6 < Vpp < VDD. We actually have a layout we're testing before we publish, so this might save you some time...

LeonMatthes, 2021-12-16:
From recent discussion on Discord it seems this is now part of the new board design.
@sqfmi can you confirm?
And has anyone tried it? Can you keep BLE active during light sleep?

This might seriously be a reason for me to get a newer version.

closed 2026-01-31 by sqfmi