sqfmi/watchy-hardware #8: [Feature Request] NFC

aceggroll, 2021-06-23:
For next revision, can you add NFC components. (eg https://www.nxp.com/docs/en/nxp/data-sheets/PN532_C1.pdf) 
Could then use Watchy with other NFC devices, eg  https://arduinogetstarted.com/tutorials/arduino-rfid-nfc-door-lock-system
Happy to be first to sign up on CrowdSupply!

danShumway, 2021-07-22:
I'll also throw out a vote for NFC. Would really open some doors for using the device in interesting ways I think, considering how cheap NFC tags are.

I'm sticking tags up around my house for triggering actions with my phone, but I don't always carry my phone. I will almost always have a watch on though, so Watchy would be just completely perfect for something like this.

adapt-L, 2024-08-04:
It would be cool if watchy had this capability. But what antenna would this require? I was looking at how Flipper Zero implemented NFC and RFID, and it seems like it might require a large circuit board to act as the antenna.

adapt-L, 2024-08-04:
But yeah, it would be really nice to use watchy as a wireless-swiss-army-knife like Flipper Zero (You can use it to clone your NFC/RFID cards). It already has the hardware to compete with products like https://dstike.com/products/dstike-deauther-watch-v3s