sqfmi/watchy-hardware #13: New PCB revision released with no documentation

mkengineering, 2021-11-25:
Recently I purchased a [Watchy from Mouser](https://www.mouser.com/ProductDetail/SQFMI/SQFMI-WATCHY-10?qs=DRkmTr78QARN9VSJRzqRxw%3D%3D) and I noticed some issues with time not updating using the example watch faces. After some debugging I noticed the PCB shown in the documentation and hardware design files differed from my board. It appears a new version has been released which has replaced the previous DS3231 RTC with a [PCF8563](https://www.nxp.com/docs/en/data-sheet/PCF8563.pdf). Unfortunately it seems this repo, the SW library and all of the user documentation has not been updated.

[image]

filipproch, 2021-11-30:
@miekush not sure if you are on Discord, so sharing here as well
[image]

sqfmi, 2021-12-25:
Sorry for the delay, pushed to v1.5

closed 2021-12-25 by sqfmi

miekush, 2021-12-25:
No worries! Thank you.