sqfmi/watchy-hardware #14: Display RST circuit should be pulled up

DarkZeros, 2021-12-18:
Hi, as part of my developments to get Watchy low power (and solar). https://github.com/sqfmi/Watchy/issues/118 
I found out that there were inconsistent power usages in the display module.

Just "touching" the connector would lead to differences in power usage from 130uA-600uA.
Also different displays seem to behave differently (V1-V2, color, ...). Agains same range 130uA-600uA.

I ended up finding that the RES# pin was the culprit, just touching that one would change the results.
RES# of the display is supposed to be left connected to HIGH.
If it is not left in that state the Dsiplay hardware will not fully enter deep sleep.

I checked that with ANY display and any watchy (I have 2), leaving this pin to 3V3 sets the power usage to 125uA.

The problem is, it can't be done on SW, since the RES pin is connected to a normal GPIO and not to a RTC GPIO.
GPIOs are left high impedance in deep sleep and therefore not driven to any value.
Solutions:
- Pullup resistor of 100k-1M (I measured no current flowing in the pin at all <1uA).
- Connect RES to RTC pin and leave the PIN as output + HIGH state.

DarkZeros, 2021-12-18:
I can confirm that a 1M resistor to 3V3 fixes the problem! (On v1.0 HW)

sqfmi, 2021-12-24:
This is great! We'll add this fix in the next HW revision!

closed 2026-01-31 by sqfmi