sqfmi/watchy-hardware #22: USB-C

MichalLebeda, 2023-04-10:
It would be great to use more durable USB-C. The downside is that it would make all the existing cases obsolete. On the other hand it is better to do sooner than later, as all the devices are migrating to type C currently.

onionhammer, 2023-06-10:
I was about to post this independently, I ordered this but definitely holding my nose about the micro USB

adapt-L, 2024-08-04:
I concur with OP. microUSB is very fragile and USB-C has quickly become the standard. Perhaps it is possible to retrofit boards by soldering something like this https://www.youtube.com/watch?v=V-vFtiDYiIw but this is a lot of extra work for the end user.

adapt-L, 2024-08-04:
> I would need to modify the device's port and maybe even the circuitry to accommodate the new connector's dimensions and functionalities. I would also like MichalLebeda mentioned need to change the case...

You wouldn't need to modify the circuit board. If you are worried about spacing, you could solder on some wires and install the new port on the other side of the board. I think it will likely require a case redesign though, which is fine because I was planning on making a custom case anyways. I have a soldering iron so I might try retrofitting it after I am done experimenting with the software.

lealife, 2025-04-22:
https://github.com/xywatch/watchy-hardware-v2 fix it

CuiYao631, 2025-05-11:
> https://github.com/xywatch/watchy-hardware-v2修复它

I’m glad to see that someone has made a USB-C version. I will try using your version.