sqfmi/watchy-hardware #23: [reality check] should pins 13 and 14 on the display connector be shorted?

shtrom, 2024-08-24:
tl;dr: should pins 13 (SCK) and 14 (MOSI) on the [display connector (J1)](https://github.com/sqfmi/watchy-hardware/blob/main/WatchySchematic.pdf) be shorted?

The buttons on my Watchy died, so I set out to replace them with Wurth Elektronik 436351045816 436351045816 from Mouser (https://github.com/sqfmi/Watchy/issues/49#issuecomment-2308150293). That worked, and I can get a program running on the watch that confirms they are functional.

But my display no longer refreshes. So I suspect I might have damaged something else when desoldering/resoldering. When testing the display connector, I noticed that pin 13 (SCK) and 14 (MOSI) are shorted. Should this be the case, or is this what I need to fix?

etwasmitbaum, 2024-08-26:
SCK and MOSI are pins needed for SPI communication, so they should not be shorted.
Since both are outputs, the GPIO pins on the ESP32 could be dead.
You could get lucky and both still work. 
Remove the short and try again.
Good luck

shtrom, 2024-08-26:
Cool, thanks for this!

They are not explicitely shorted, but when I test them, there is 0 ohm between them; so, a short.

At least I'm on the right investigative path. I'll see how I can test those pins.

Thanks!

shtrom, 2024-10-06:
Ok, more on this. I think I was counting wrong, and was instead looking at pins 15 and 16 which... are both +3V3, so it makes sense they measure as a short.

[image]

closed 2024-10-06 by shtrom

shtrom, 2024-10-21:
Conclusion: I ordered a replacement eInk display from Good Display, and that worked. My issue was that I somehow borked the display.